STMicroelectronics/STM32CubeG4 #1: MDK-ARM startup assembly files have invalid CMSIS-Pack/Config Wizard Annotations

jkrech, 2019-05-09:
Please review all MDK-ARM startup_*.s files. Note that these files follow the "CMSIS Config Wizard Annotations" (see: http://arm-software.github.io/CMSIS_5/Pack/html/configWizard.html)

The files contain this code in the header section.

`  ;* <h2><center>&copy; Copyright (c) 2017 STMicroelectronics.`
`  ;* All rights reserved.</center></h2>`

Which confuses the config annotation parser. Therefore I suggest to move the start tag of the config wizard annotation 
`
;* <<< Use Configuration Wizard in Context Menu >>>
`
to after the copyright and license (line 27)

` ;******************************************************************************* `
` ;* <<< Use Configuration Wizard in Context Menu >>> `
` ; Amount of memory (in bytes) allocated for Stack`

ALABSTM, 2019-05-09:
Dear Joachim Krech,

Thank you for pointing out this issue. It is now being analyzed. We will provide you with feedback as soon as possible.

With regards

ALABSTM, 2019-05-20:
Dear Joachim Krech,

Thank you once more for the issue you pointed out. The solution you suggested will be adopted and made available within a future release of the STM32CubeG4 firmware.

With regards

ALABSTM, 2019-05-20:
ST Internal Reference: 66667

closed 2019-06-03 by ALABSTM