STMicroelectronics/STM32CubeG4 #2: STM32Cube_FW_G4_V1.1.0 HRTIM_Buck_Sync_Rect cannot not directly import

VictorTagayun, 2019-10-14:
Hello,
Newbie here, I was trying to import HRTIM_Buck_Sync_Rect found in 

STM32Cube_FW_G4_V1.1.0\Projects\B-G474E-DPOW1\Examples_MIX\HRTIM\

When I import the folder, I would need to generate code first, but will not be able to build.

So I go Project Manager >> Code Generator >> Copy only the necessary files from "Add necessary files ...."

Then generate code and I will be able to build but with errors on using LL functions and insert library and BSP codes.

I can see that in the HRTIM, the init was done using HAL but inside the HRTIM interrupt and Main funtions, LL was used.

Can HAL and LL of the same peripherals use both HAL and LL?
How to link the LL library in the project?
How to add link to BSP api and codes?
How to import the project files?

I am using STMCudeIDE 1.0.2 on windows 7. I tried on Linux, almost same errors can be seen.

ALABSTM, 2019-10-16:
Hi Victor,

Glad to receive your message. Unfortunately, issues are only intended to the firmware published within the different repositories of this GitHub organization.

For subjects such as yours related to CubeMX, please submit your question to the ST Community under the ["STM32CubeMX"](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) topic.

Thank you once more for your message and hopefully you get the answers you are looking for from the Community.

With regards,

ALABSTM, 2019-10-16:
Hi Victor,

Allow me to close this ticket as it is out of the scope of this GitHub organization.

With regards,

closed 2019-10-16 by ALABSTM

VictorTagayun, 2019-10-17:
Did you make and compile the folder as project before you upload and/or release?
Maybe it should be compiled first so anyone can import it immediately.