STMicroelectronics/STM32CubeG4 #4: LL SPI_TwoBoards_FullDuplex_DMA_Master_Init examples

phryniszak, 2020-01-28:
In SPI_TwoBoards_FullDuplex_DMA_Master_Init examples
SPI1_RX is connected to DMA1_Channel1 and  SPI1_TX is connected to DMA1_Channel3,
but in code DMA channel 3 is reporting DMA1_ReceiveComplete_Callback and DMA channel 1 DMA1_TransmitComplete_Callback.

The same here:
`LL_DMA_SetDataLength(DMA1, LL_DMA_CHANNEL_3, ubNbDataToReceive);`
`LL_DMA_SetDataLength(DMA1, LL_DMA_CHANNEL_1, ubNbDataToTransmit);`

To  set transmit size we use ubNbDataToReceive and to receive ubNbDataToTransmit.

Examples are executing ok but naming is at least confusing.

ASELSTM, 2020-02-03:
Hi Phryniszak,

Thank you for your contribution. This issue has been forwarded to our development teams. We will be back to you as soon as they provide us with feedback.

With regards,

ASELSTM, 2020-02-04:
Hi Phryniszak,

Our development teams confirmed the issue you pointed out. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-02-04:
ST Internal Reference: 80465

closed 2020-02-07 by ALABSTM

ALABSTM, 2020-03-04:
Hi Phryniszak,

This issue has been fixed in the frame of the STM32CubeG4 **v1.2.0** firmware package already available on GitHub.

With regards,