STMicroelectronics/STM32CubeG4 #5: UART baud rate is double when configuring PLLs using LL drivers on STM32G4xxx.

intercreate, 2020-02-07:
**Describe the set-up**
 * [NUCLEO_G474RE](https://www.st.com/en/evaluation-tools/nucleo-g474re.html)
 * RTOS running is [Zephyr](https://www.zephyrproject.org/)
 * The environment uses the Zephyr SDK for compiler, cmake / Ninja for build. all command line BASH.

**Describe the bug**
The function `LL_PLL_ConfigSystemClock_HSI()` in `stm32g4xx_ll_utils.c` temporarily divides the AHB clock by 2 if the SYSCLK divider is 1. According to the comments, this is to "Prevent undershoot at highest frequency by applying intermediate AHB prescaler 2".

The issue is that during this temporary AHB clock rate, the `SystemCoreClock` is updated to also be 1/2 the expected clock.

Although `LL_PLL_ConfigSystemClock_HSI()` resets the AHB clock properly, it does not reset the global variable `SystemCoreClock`.

This incorrect value is exposed when trying to set the UART baud rate later on in `stm32g4xx_hal_uart.c`.

Note that Zephyr uses the latest ([0f8a4fd](https://github.com/STMicroelectronics/STM32CubeG4/commit/0f8a4fd5d6bf5853a1aea530a738ee55abbb4c0f)) with [this](https://github.com/STMicroelectronics/STM32CubeG4/pull/3) recent pull-request integrated.

**How To Reproduce**
1. Configure SYSCLK to use the PLL. Divide it by 1 so it stays above 80MHz. Use `LL_PLL_ConfigSystemClock_HSI()` to configure PLLs using the pull-request mentioned above.
2. Use the UART HAL to configure an interface with baud rate 115200. Actual baud will be double.

See https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/clock_control/clock_stm32_ll_common.c#L346 for use of `LL_PLL_ConfigSystemClock_HSI()`

**Additional context**

This patch fixes the problem in my environment.

```
diff --git a/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_utils.c b/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_utils.c
index ea78949a..347d19b4 100644
--- a/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_utils.c
+++ b/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_utils.c
@@ -329,6 +329,7 @@ ErrorStatus LL_PLL_ConfigSystemClock_HSI(LL_UTILS_PLLInitTypeDef *UTILS_PLLInitS
     {
       UTILS_ClkInitStruct->AHBCLKDivider = LL_RCC_SYSCLK_DIV_1;
       LL_RCC_SetAHBPrescaler(UTILS_ClkInitStruct->AHBCLKDivider);
+      LL_SetSystemCoreClock(pllfreq);
     }
   }
   else
@@ -420,6 +421,7 @@ ErrorStatus LL_PLL_ConfigSystemClock_HSE(uint32_t HSEFrequency, uint32_t HSEBypa
     {
       UTILS_ClkInitStruct->AHBCLKDivider = LL_RCC_SYSCLK_DIV_1;
       LL_RCC_SetAHBPrescaler(UTILS_ClkInitStruct->AHBCLKDivider);
+      LL_SetSystemCoreClock(pllfreq);
     }
   }
   else
```

**Screenshots**
N/A

ALABSTM, 2020-02-12:
Hi InterCreate,

Thank you for having created this issue. It will be forwarded to our development teams for deeper analysis. I also add @FRASTM into the loop as he has reported the same issue via this [pull-request](https://github.com/STMicroelectronics/STM32CubeG4/pull/3).

I also indicate the [link](https://community.st.com/s/question/0D50X0000C5TOy8SQG/uart-baud-rate-is-double-when-configuring-plls-using-ll-drivers-on-stm32g4xxx) to your post for the ST Community.

I will be back to you as soon as I get an answer. Thank you for your patience and thank you both, you and @FRASTM, once more, for having reported the issue.

With regards,

ALABSTM, 2020-03-04:
ST Internal Reference: 79947

ALABSTM, 2020-03-04:
Hi InterCreate,

This issue has been fixed. The fix is available in the frame of the STM32CubeG4 **v1.2.0** which has been recently published on GitHub.

Thank you for your patience and thank you both (@intercreate and @FRASTM) for your contribution. This issue can be closed now.

With regards,

closed 2020-03-04 by ALABSTM