STMicroelectronics/STM32CubeG4 #6: HAL_GPIO doesn't have EXTI enable / disable API.

suikan4github, 2020-03-10:
The EXTI interrupt is caught by the HAL_GPIO_EXTI_Callback(). But there is not HAL_GPIO API to enable / disable EXTI. 

This is not appropriate design. Some API to enable / disable EXTI for each line should be provided.

ALABSTM, 2020-03-13:
Hi Suikan4GitHub,

First of all, thank you for asking the question.

Actually, the EXTI enabling and disabling are handled at HAL_GPIO_Init() and HAL_GPIO_DeInit() levels respectively. I hope you find in the explanations below a satisfying answer to your question.

As for the EXTI enabling, let's examine the GPIO_EXTI example:
* At user application level, GPIO mode is set to "GPIO_MODE_IT_FALLING" (as shown below).
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Projects/NUCLEO-G474RE/Examples/GPIO/GPIO_EXTI/Src/main.c#L179
* Constant "GPIO_MODE_IT_FALLING" is defined at driver's level (see below).
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal_gpio.h#L125
* Actually, it is a collection of **ORed** constants:
  * GPIO_MODE_IT_FALLING = EXTI_MODE | FALLING_EDGE | GPIO_MODE_IT
  * These constants are also defined at drivers' level, as private ones, however (below an example).
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c#L130
* Finally, still at driver's level, within the HAL_GPIO_Init() function, EXTI line is configured.
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c#L246
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c#L280
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_gpio.c#L255

With regards,

suikan4github, 2020-03-14:
HI ALABSTM

Thank you very much for your explanation. 

I still have difficulties to use the HAL_GPIO_Init() and HAL_GPIO_DeInit() as enable / disable the EXTI. The problem is the GPIO_Init parameter of the HAL_GPIO_Init(). This parameter is made by CubeIDE but discarded when the program returns from MX_GPIO_Init(). As a result, the application cannot use the initialization parameter ( Similar parameter exists for the other peripherals, as Init member variable of the peripheral handle ). 

So, from the application point of view, explicit enable/disable API is explicit and useful. 

Regards,
Seiichi

ALABSTM, 2020-04-02:
Hi Seiichi,

I hope you are doing well. Depending on the use case you are thinking about, you point might be relevant. However, in this case, it is no more related to the firmware published within this repository. It is rather related to **Cube MX** or even way you think to design you application.

In both cases, please allow me to redirect you to the ST Community:
* For aspects related to Cube MX, please submit your questions or issues [here](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx).
* For aspects related toGPIO and EXTI, please submit your questions or issues [here](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus).

Anyway, about the "GPIO_InitStructure" declared locally, may I suggest something? I will use the [GPIO_EXTI](https://github.com/STMicroelectronics/STM32CubeG4/tree/master/Projects/NUCLEO-G431RB/Examples/GPIO/GPIO_EXTI) as example.
* In the user dedicated section, declare a new variable of type "GPIO_InitTypeDef".
https://github.com/STMicroelectronics/STM32CubeG4/blob/546320ee7624214ada13cef434b715926dd00978/Projects/NUCLEO-G431RB/Examples/GPIO/GPIO_EXTI/Src/main.c#L48-L49
* Inside the "EXTI15_10_IRQHandler_Config()" function, copy the local structure into the one you declared (just after call to "HAL_GPIO_Init()" for instance).
https://github.com/STMicroelectronics/STM32CubeG4/blob/546320ee7624214ada13cef434b715926dd00978/Projects/NUCLEO-G431RB/Examples/GPIO/GPIO_EXTI/Src/main.c#L184

I hope this hint is helpful. Do not hesitate to visit the [ST Community Forum](https://community.st.com/s/). There, you will also find support and answers to your questions hopefully.

With regards,

suikan4github, 2020-04-03:
Hi ALABSTM

Thank you for your kind reply. I will think of some other way to disable/enable interrupt. 

Regards

closed 2020-04-03 by suikan4github