STMicroelectronics/STM32CubeG4 #7: NUCLEO-G431RB/Examples/PWR/PWR_SHUTDOWN doesn't work

jolo1581, 2020-03-29:
**Describe the set-up**
 * Using Nucleoboard NUCLEO-G431RB
 * IAR Embedded Workbench 8.4, IAR Compiler 8.4

**Describe the bug**
Compiling and downloading this example, the controller won't go to the shutdown mode. After pressing User Button, the controller starts up again. Using ST-Link, in Shutdown mode it must loose connection to target, but debugging runs.

**How To Reproduce**
1. Compiling and download software from NUCLEO-G431RB/Examples/PWR/PWR_SHUTDOWN
2. Press user button
3. Controller doen't shut down like described in readme.txt

ASELSTM, 2020-04-09:
Hi @jolo1581 

Thank you for your contribution.  We haven’t been able to reproduce the issue you describe. 

We have tested the example, the MCU enters in shutdown mode when pressing the user button and the connection with the debugger is then lost. Therefore, no debug task could then be performed even when system resume from shutdown mode.

Take care and stay safe.
Best regards,

jolo1581, 2020-04-09:
@ASELSTM maybe it is a problem using IAR Compiler with EWARM 8.4?

ASELSTM, 2020-04-10:
Hi @jolo1581,

We have also tested with IAR Embedded Workbench 8.4 and with the same IAR Compiler 8.4.

Could you please check those steps to better understand the source of your problem :

1. Compile and download the software on the NUCLEO-G431RB board
2. Press user button
3. Press debugger stop button
4. Check whether you have communication error messages as shown on the image above.

     [image]

Best regards,

jolo1581, 2020-04-15:
I tried the steps above.

When I only press the user button short, it doesn't work.
If I press and hold user button, the device lost debug communication.

But the readme says:

> When the User push-button is pressed a falling edge is detected on the EXTI line and
an interrupt is generated. In the EXTI handler routine, the wake-up pin PWR_WAKEUP_PIN2 
is enabled and the corresponding wake-up flag cleared. 
Then, the system enters SHUTDOWN mode causing LED2 to stop toggling. 
Next, the user can wake-up the system in pressing the User push-button which is connected
to the wake-up pin PWR_WAKEUP_PIN2.
A falling edge on the wake-up pin wakes-up the system from SHUTDOWN.
Alternatively, an external RESET of the board leads to a system wake-up as well.

ASELSTM, 2020-04-16:
Hi @jolo1581,

May I ask you the STM32CubeG4 firmware revision you are using ? 

Best regards,

jolo1581, 2020-04-16:
I tried with latest master

ASELSTM, 2020-04-17:
Hi @jolo1581 

There is indeed a discrepancy between the description of the README and the behavior indeed.

The problem is not coming from the shutdown mode entry, the system has entered this mode when the button is pressed, when the falling edge is detected on the EXTI line.

However, it is the shutdown mode exit scenario that is different from the README.
Indeed, according to the README the system must be waked up when user pushes the button a second time, but the system exits this mode when the button is released.

An internal bug tracker will be opened and the correction will be available in a future release. 

Thank you once more for your contribution.

Best regards,

ASELSTM, 2020-04-17:
ST Internal Reference: 84841

jolo1581, 2020-04-17:
@ASELSTM thanks for feedback.

jolo1581, 2020-06-03:
Hello,

I did some work with STM32G431RB and shutdown mode.
My wakeup souce pin PA2 could be high when going down to shutdown. When this happens, the controller starts up again.

I did my shutdown like in example and identify the problem of wakeup when pin is high while calling ` HAL_PWREx_EnterSHUTDOWNMode();`

[image]

When I removed the marked code ` HAL_PWR_DisableWakeUpPin(PWR_WAKEUP_PIN2);` and `HAL_PWR_EnableWakeUpPin(PWR_WAKEUP_PIN2_LOW);`

And added `HAL_PWR_EnableWakeUpPin(PWR_WAKEUP_PIN2_LOW); `to initialisation in main loop the shutdown works right in my custom setup.

Maybe it helps you to identify the probem.

ASELSTM, 2021-08-13:
Hi @jolo1581,

Since the issue you pointed out has been fixed, please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2021-08-13 by ASELSTM