STMicroelectronics/STM32CubeG4 #8: Release note issue ?

fpistm, 2020-04-30:
Hi,

in the release note of the CubeG4 v1.2.0, it is mentioned:
https://github.com/STMicroelectronics/STM32CubeG4/blob/21b2f2fa52ef813c38285abb6b04636910d367c9/Release_Notes.html#L133-L137

But I could not find any code related to `HAL_I2C_WRONG_START`?

Same in the G4 HAL drivers release 1.1.1:
https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/be4da93755336778a77b65ae5d85e97070b82750/Release_Notes.html#L118-L122

ALABSTM, 2020-04-30:
Hi @fpistm,

Thank you for your report. It will be forwarded to our development teams for analysis. We will be back to you as soon as they provide us with their feedback.

With regards,

ALABSTM, 2020-04-30:
Hi @fpistm,

Our development teams confirmed the inconsistency you pointed out. Actually, the mentioned update concerns only STM32F1, F2, F4, and L1 series. As for the STM32G4 series, this is an error in the release note that shall be corrected next release, hopefully.

Thank you again for having reported this point. Take care and stay safe.

With regards,

ALABSTM, 2020-04-30:
ST Internal Reference: 85698

closed 2020-07-29 by ALABSTM