STMicroelectronics/STM32CubeG4 #9: PCAN Timeout after a long (days) run.

paoloteti, 2020-05-04:
**Describe the set-up**
 * Custom board based on STM32G431
 * GNU arm-none-eabi 7-2018-q2-update

**Describe the bug**
PCAN Timeout after a long (days) run.

**How To Reproduce**
1. Indicate the global behavior of your application project.
No PCAN Timeout 

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
FDCAN driver (stm32g4xx_hal_fdcan.c)

3. The use case that generates the problem.
Long (days) run.

4. How we can reproduce the problem.

**Additional context**
In this piece of code if `tickstart` is 0xFFFFFFFF and a Tick ISR occur in the middle and so the next call to `HAL_GetTick()` returns 0 the resulting timeout check return a timeout event.

```
  tickstart = HAL_GetTick();

  /* Check Sleep mode acknowledge */
  while ((hfdcan->Instance->CCCR & FDCAN_CCCR_CSA) == FDCAN_CCCR_CSA)
  {
    if ((HAL_GetTick() - tickstart) > FDCAN_TIMEOUT_VALUE)

```

ALABSTM, 2020-05-06:
Hi @paoloteti,

Thank you for this detailed report. it has been forwarded to our development teams. I will be back to you as soon as they provide me with feedback.

Thank you once more. Take care and stay safe.

With regards,

ALABSTM, 2020-05-08:
Hi @paoloteti,

According to our development teams, the case you are describing does not lead to a timeout. Indeed, if:
* `tickstart` = 0xFFFFFFFF
* `HAL_GetTick()` returns 0

The difference would give (0x00000000 - 0xFFFFFFFF) = 0x00000000 + 0x00000001 = 0x00000001
Where 0x00000001 is the two's complement of 0xFFFFFFFF.

The result is not greater than `FDCAN_TIMEOUT_VALUE`. Thus no timeout occurs.

In case you had a timeout, the cause must be something else.
* Would you mind checking again from your side whether you can reproduce the issue again?
* In case you can, may I ask you the frequency of appearance of the issue?

Thank you,

closed 2020-05-08 by paoloteti

ALABSTM, 2020-05-08:
ST Internal Reference: 66832