STMicroelectronics/STM32CubeG4 #10: The value passed to LL_RCC_SetAPB2Prescaler is incorrect

fivdi, 2020-05-23:
**Describe the set-up**
 * Board: NUCLEO-G431RB
 * Compiler and its version: arm-none-eabi-gcc (xPack GNU Arm Embedded GCC, 32-bit) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599].

**Describe the bug**
`LL_RCC_SetAPB2Prescaler` is called in [Projects/NUCLEO-G431RB/Examples_LL/UTILS/UTILS_ConfigureSystemClock/Src/main.c](https://github.com/STMicroelectronics/STM32CubeG4/blob/v1.2.0/Projects/NUCLEO-G431RB/Examples_LL/UTILS/UTILS_ConfigureSystemClock/Src/main.c) to set the APB2 prescaler on [this line](https://github.com/STMicroelectronics/STM32CubeG4/blob/v1.2.0/Projects/NUCLEO-G431RB/Examples_LL/UTILS/UTILS_ConfigureSystemClock/Src/main.c#L191):
```c
  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
```
The value passed to `LL_RCC_SetAPB2Prescaler` is `LL_RCC_APB1_DIV_1` which is incorrect. The correct value to pass to `LL_RCC_SetAPB2Prescaler` is `LL_RCC_APB2_DIV_1`.

At the end of the day this doesn't actually cause any problems because both `LL_RCC_APB1_DIV_1` and `LL_RCC_APB2_DIV_1` are defied as 0. However, it would still make sense to use `LL_RCC_APB2_DIV_1` rather than `LL_RCC_APB1_DIV_1` when appropriate.

**Additional context**
There are several other occurrences of this issue in the code.

```
$ grep -rn LL_RCC_SetAPB2Prescaler | grep LL_RCC_APB1_DIV_1
NUCLEO-G474RE/Examples_LL/ADC/ADC_SingleConversion_TriggerSW_IT_Init/Src/main.c:260:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/ADC/ADC_AnalogWatchdog_Init/Src/main.c:270:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/ADC/ADC_ContinuousConversion_TriggerSW_Init/Src/main.c:323:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/ADC/ADC_SingleConversion_TriggerSW_Init/Src/main.c:318:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/ADC/ADC_GroupsRegularInjected_Init/Src/main.c:317:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/ADC/ADC_Oversampling_Init/Src/main.c:386:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/DMA/DMA_CopyFromFlashToMemory_Init/Src/main.c:207:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/PWR/PWR_EnterStandbyMode/Src/main.c:183:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/PWR/PWR_EnterStopMode/Src/main.c:182:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/LPTIM/LPTIM_PulseCounter_Init/Src/main.c:170:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/SPI/SPI_TwoBoards_FullDuplex_DMA_Slave_Init/Src/main.c:233:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/SPI/SPI_OneBoard_HalfDuplex_IT_Init/Src/main.c:213:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/SPI/SPI_TwoBoards_FullDuplex_DMA_Master_Init/Src/main.c:239:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/WWDG/WWDG_RefreshUntilUserEvent_Init/Src/main.c:180:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RNG/RNG_GenerateRandomNumbers/Src/main.c:188:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RNG/RNG_GenerateRandomNumbers_IT/Src/main.c:193:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/LPUART/LPUART_WakeUpFromStop_Init/Src/main.c:194:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/LPUART/LPUART_WakeUpFromStop/Src/main.c:204:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/CORDIC/CORDIC_CosSin/Src/main.c:212:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/COMP/COMP_CompareGpioVsVrefInt_OutputGpio_Init/Src/main.c:193:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/COMP/COMP_CompareGpioVsVrefInt_IT_Init/Src/main.c:181:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/GPIO/GPIO_InfiniteLedToggling_Init/Src/main.c:165:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_Communication_Rx_IT_Continuous_Init/Src/main.c:206:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_Communication_Tx_Init/Src/main.c:193:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_WakeUpFromStop_Init/Src/main.c:200:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_Communication_Tx_IT_Init/Src/main.c:182:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_Communication_Rx_IT_Init/Src/main.c:177:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/USART/USART_Communication_TxRx_DMA_Init/Src/main.c:195:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RTC/RTC_ExitStandbyWithWakeUpTimer_Init/Src/main.c:220:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RTC/RTC_TimeStamp_Init/Src/main.c:272:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RTC/RTC_ProgrammingTheWakeUpTimer/Src/main.c:222:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RTC/RTC_Alarm_Init/Src/main.c:275:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RTC/RTC_Tamper_Init/Src/main.c:221:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/CRC/CRC_UserDefinedPolynomial/Src/main.c:199:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/CRC/CRC_CalculateAndCheck/Src/main.c:191:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_OneBoard_Communication_PollingAndIT_Init/Src/main.c:205:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_OneBoard_AdvCommunication_DMAAndIT_Init/Src/main.c:273:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_TwoBoards_MasterTx_SlaveRx_DMA_Init/Src/main.c:235:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_TwoBoards_MasterRx_SlaveTx_IT_Init/Src/main.c:207:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_OneBoard_Communication_IT_Init/Src/main.c:194:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_TwoBoards_MasterTx_SlaveRx_Init/Src/main.c:240:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_TwoBoards_WakeUpFromStop_IT_Init/Src/main.c:203:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/I2C/I2C_OneBoard_Communication_DMAAndIT_Init/Src/main.c:218:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_DMA_Init/Src/main.c:231:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_BreakAndDeadtime_Init/Src/main.c:177:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_OnePulse_Init/Src/main.c:177:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_InputCapture_Init/Src/main.c:229:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_PWMOutput_Init/Src/main.c:216:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/TIM/TIM_OutputCompare_Init/Src/main.c:206:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/EXTI/EXTI_ToggleLedOnIT/Src/main.c:160:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/EXTI/EXTI_ToggleLedOnIT_Init/Src/main.c:160:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G474RE/Examples_LL/RCC/RCC_UseHSI_PLLasSystemClock/Src/main.c:226:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
[... 53 lines cut ...]
NUCLEO-G431RB/Examples_LL/I2C/I2C_OneBoard_Communication_IT_Init/Src/main.c:194:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/I2C/I2C_TwoBoards_MasterTx_SlaveRx_Init/Src/main.c:240:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/I2C/I2C_TwoBoards_WakeUpFromStop_IT_Init/Src/main.c:203:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/I2C/I2C_OneBoard_Communication_DMAAndIT_Init/Src/main.c:218:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/TIM/TIM_DMA_Init/Src/main.c:231:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/TIM/TIM_BreakAndDeadtime_Init/Src/main.c:177:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/TIM/TIM_OnePulse_Init/Src/main.c:177:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
[... cut ...]
NUCLEO-G431RB/Examples_LL/DAC/DAC_GenerateConstantSignal_TriggerSW_Init/Src/main.c:198:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
NUCLEO-G431RB/Examples_LL/DAC/DAC_GenerateConstantSignal_TriggerSW_LP_Init/Src/main.c:227:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
B-G474E-DPOW1/Applications/USB-PD/USB-PD_Consumer_1port/Src/main.c:199:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
STM32G474E-EVAL/Applications/USB-PD/USB-PD_Provider_1port/Src/main.c:199:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
STM32G474E-EVAL/Applications/USB-PD/USB-PD_Consumer_1port/Src/main.c:199:  LL_RCC_SetAPB2Prescaler(LL_RCC_APB1_DIV_1);
```

RKOUSTM, 2020-05-27:
Hi @fivdi,

Thank you for having reported this point. Your remark will be transmitted to our development teams to have this update in the frame of a future release. 

With regards,

RKOUSTM, 2020-05-27:
Hi @fivdi,

Thank you for this report, once again.  This issue has been reported to the CubeMX team. A fix will be available in a future CubeMX release. 

As this is a CubeMX-related issue and not directly related to the firmware published in this repository, please allow me to close it. Thank you for your comprehension and contribution.

With regards,

RKOUSTM, 2020-05-27:
ST Internal Reference: 87089

closed 2020-05-27 by RKOUSTM