STMicroelectronics/STM32CubeG4 #12: HRTIM Fault Source Settings using LL Drivers

hanneskiene, 2020-06-29:
**Describe the set-up**
 * STM32G474RE
 * STMCubeIDE under Linux using the provided g++ toolchain

**Describe the bug**
The bug is in the low level driver for the HRTIM peripheral.
Setting the Fault Source Register using LL_HRTIM_FLT_SetSrc(...) uses a wrong mask
value to modify the contents of the second fault source register.
This results in previous settings to be overridden in the upper half of the register.
Please see file stm32g4xx_ll_hrtim.h, Line 9658 (or the previous mask generation statement).

**How To Reproduce**
1. Use the CubeMX Package to configure the HRTIM peripheral using multiple Fault Inputs with different fault sources.
Especially Faults whose fault source bits are located in the upper half of the second fault source register (e.g. Fault4).
2. Run the generated code to just before LL_HRTIM_FLT_SetSrc(..) for the second configured channel(Fault5).
The fault Source bits for Fault4 are set correctly.
3. Execute the second LL_HRTIM_FLT_SetSrc(..)
This statement clears the upper half of the second fault source register -> Disables the fault4 source setting.

A patch would include a corrected Mask value for the second Fault Source Register.

ALABSTM, 2020-08-28:
Hi @hanneskiene,

Thank you for this contribution. The issue you pointed out has been confirmed by our development teams. I will get back to you as soon as they provide feedback about the fix to be implemented. Thank you once more.

With regards,

ALABSTM, 2020-08-28:
ST Internal Reference: 91816

closed 2021-03-08 by ALABSTM