STMicroelectronics/STM32CubeG4 #15: Unable to compile USB PD middleware with USBCORE_VPD config

intercreate, 2020-10-06:
**Describe the set-up**
 * STM32G474-EVAL hardware
 *  `Projects/STM32G474E-EVAL/Demonstrations/Modules/ucpd/` project with `-DUSBCORE_VPD=1` defined on **gcc** command line.
 * stm32cube IDE
 * Project using `USBPDCORE_PD3_FULL_CM4_wc32.a`

**Describe the bug**
Besides a few compile errors with the source code, there's a linker error for `USBPD_PE_StateMachine_SNKwVPD()` implementation. This should be included in the library, but the linker cannot find it.

**How To Reproduce**
Compile `Projects/STM32G474E-EVAL/Demonstrations/Modules/ucpd/` with `-DUSBCORE_VPD=1` to enable the VPD code.

**Additional context**
USBCORE_VPD is not defined anywhere in `usbpd_def.h`. Does this mean the STCube library does not support VPD / VPA as defined in the USB PD specification?

ALABSTM, 2020-10-14:
Hi @intercreate,

Thank you for your report. We have forwarded your question to our development teams. We will get back to you as soon as we have their feedback.

With regards,

NPESTM, 2020-12-11:
Hello,
VPD development has started on our side but was put on hold, because we have no way to test the feature with official testers.
So the feature cannot be delivered externally today as it is. But as soon as we have ways to validate it will we go on.
So current status is : not supported for the moment, but topic is being followed.

ALABSTM, 2022-02-14:
Hi @intercreate,

We discussed further your issue with our development teams and, according to them, there is **no plan** to support such feature for the moment. Please allow me to close the issue. Thank you for your comprehension.

With regards,

closed 2022-02-14 by ALABSTM