STMicroelectronics/STM32CubeG4 #17: Feature request: HRTIM add option to disable fault blanking window

ldoesvictron, 2020-10-12:
As of STM32CubeG4 v1.3.0 it is not possible to generate the HRTIM fault lines initialization code without it also calling to HAL_HRTIM_FaultBlankingConfigAndEnable(). Please add the option for the blanking window to be either disabled or enabled to the Fault Lines Configuration window in CubeMX.

ALABSTM, 2021-02-24:
Hi @ldoesvictron,

If I understand correctly, you would like to configure HRTIM's fault lines without enabling the blanking. The issue you are facing is that with Cube MX, function `HAL_HRTIM_FaultBlankingConfigAndEnable()` is called from the `MX_HRTIM1_Init()` which **automatically enables** the blanking (as pointed out in the note below).
https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_hrtim.c#L4184

Actually, this is rather a HAL-related point than a Cube MX- one. As a solution, may I suggest the following:
* At the end of the `MX_HRTIM1_Init()` function, you would notice two user tags in the form of comments (see code snippet below).
* Simply use the macro dedicated to disabling the blanking on a specified fault line (`HRTIM_FAULT_1` is taken as example in the code snippet below).
https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal_hrtim.h#L4581

```diff
  /* USER CODE BEGIN HRTIM1_Init 2 */
+  __HAL_HRTIM_FAULT_BLANKING_DISABLE(&hhrtim1, HRTIM_FAULT_1);
  /* USER CODE END HRTIM1_Init 2 */

```

I hope this helps. Do not hesitate otherwise.

With regards,

ALABSTM, 2021-05-24:
Hi @ldoesvictron,

I hope you are fine. Please allow me to close this issue as not reply has been received from your side since a couple of months. Do not hesitate to reopen it if you think it is still relevant.

With regards,

closed 2021-05-24 by ALABSTM