STMicroelectronics/STM32CubeG4 #19: G491 and G4A1 wrong RAM size in linker script

atsju, 2020-12-23:
In the linker scripts for [G491](https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Drivers/CMSIS/Device/ST/STM32G4xx/Source/Templates/iar/linker/stm32g491xx_flash.icf#L10) and [G4A1](https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Drivers/CMSIS/Device/ST/STM32G4xx/Source/Templates/iar/linker/stm32g4a1xx_flash.icf#L10) the RAM size seems to be 128kB but on [G4 series web page](https://www.st.com/content/st_com/en/products/microcontrollers-microprocessors/stm32-32-bit-arm-cortex-mcus/stm32-mainstream-mcus/stm32g4-series/stm32g4x1.html) statement is 112kB. One of the two is false.

RKOUSTM, 2020-12-24:
Hi @atsju, 

Thank you for your contribution. The reported issue has been confirmed and a bug tracker has been opened internally. The following fix in the flash `files .icf` will be implemented and made available in future release for all ``STM32G4 series``, where the RAM size will be the sum between `SRAM1` and `SRAM2`. 

- For  `STM32G4` Series `category 3` :
```patch
@@ -9,4 +9,4 @@ 
+ define symbol __ICFEDIT_region_RAM_start__     = 0x20000000;
- define symbol __ICFEDIT_region_RAM_end__       = 0x2001FFFF;
+ define symbol __ICFEDIT_region_RAM_end__       = 0x20017FFF;
+ define symbol __ICFEDIT_region_CCMSRAM_start__ = 0x10000000;
+ define symbol __ICFEDIT_region_CCMSRAM_end__   = 0x10007FFF;
```
- For  `STM32G4` Series `category 4` :
```patch
@@ -9,4 +9,4 @@ 
+ define symbol __ICFEDIT_region_RAM_start__     = 0x20000000;
- define symbol __ICFEDIT_region_RAM_end__       = 0x2001FFFF;
+ define symbol __ICFEDIT_region_RAM_end__       = 0x20017FFF;
+ define symbol __ICFEDIT_region_CCMSRAM_start__ = 0x10000000;
+ define symbol __ICFEDIT_region_CCMSRAM_end__   = 0x10003FFF;
```
- For  `STM32G4` Series `category 2` :
```patch
@@ -9,4 +9,4 @@ 
+ define symbol __ICFEDIT_region_RAM_start__     = 0x20000000;
- define symbol __ICFEDIT_region_RAM_end__       = 0x20007FFF;
+ define symbol __ICFEDIT_region_RAM_end__       = 0x200057FF;
+ define symbol __ICFEDIT_region_CCMSRAM_start__ = 0x10000000;
+ define symbol __ICFEDIT_region_CCMSRAM_end__   = 0x100027FF;
```
Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-12-24:
ST Internal Reference: 98738

ASELSTM, 2021-03-03:
Hi @atsju,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.4.0 of the STM32CubeG4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM