STMicroelectronics/STM32CubeG4 #20: [LL_USART_Init] - bad assert_param for BRR register maximum value

scachemaille, 2021-01-15:
https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_ll_usart.c#L303

this check here is too late and useless.
Because LL_USART_SetBaudrate called just above will never set BRR above 0xFFFF (as the higher bits are reserved) without any check if the calculation results in a higher result.

ASELSTM, 2021-01-25:
Hi @scachemaille,

Thank you for rising the concern. The issue you pointed out has been confirmed. Indeed the check is useless the  **IS_LL_USART_BRR_MAX** macro will be removed,  this will be logged into our internal database to be fixed.  The fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-01-25:
ST Internal Reference: 100037

ASELSTM, 2021-03-03:
Hi @scachemaille,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.4.0 of the STM32CubeG4 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-03 by ASELSTM