STMicroelectronics/STM32CubeG4 #22: Some RTC Macro did'nt use pHandle

CanastraRF, 2021-02-06:
The following RTC Macro didn't use pHandle

G4

	__HAL_RTC_WRITEPROTECTION_DISABLE( pHandle );
	__HAL_RTC_WRITEPROTECTION_ENABLE( pHandle );
	__HAL_RTC_ALARM_ENABLE_IT(pHandle, RTC_IT_ALRA);
	__HAL_RTC_ALARM_DISABLE_IT(pHandle, RTC_IT_ALRA);
	__HAL_RTC_ALARM_GET_IT_SOURCE(pHandle, RTC_IT_ALRA)

	__HAL_RTC_ALARMA_ENABLE( pHandle );
	__HAL_RTC_ALARMA_DISABLE( pHandle );
	__HAL_RTC_ALARMB_ENABLE( pHandle );
	__HAL_RTC_ALARMB_DISABLE( pHandle );

	__HAL_RTC_WAKEUPTIMER_ENABLE( pHandle );
	__HAL_RTC_WAKEUPTIMER_DISABLE( pHandle );
	__HAL_RTC_WAKEUPTIMER_CLEAR_FLAG(pHandle, RTC_FLAG_WUTF);
	__HAL_RTC_WAKEUPTIMER_ENABLE_IT(pHandle, RTC_IT_WUT);
	__HAL_RTC_WAKEUPTIMER_DISABLE_IT(pHandle, RTC_IT_WUT);
	__HAL_RTC_WAKEUPTIMER_GET_IT_SOURCE(pHandle, RTC_IT_WUT)

	__HAL_RTC_TIMESTAMP_ENABLE( pHandle );
	__HAL_RTC_TIMESTAMP_DISABLE( pHandle );
	__HAL_RTC_TIMESTAMP_CLEAR_FLAG(pHandle, RTC_FLAG_WUTF);
	__HAL_RTC_TIMESTAMP_ENABLE_IT(pHandle, RTC_IT_TS);
	__HAL_RTC_TIMESTAMP_DISABLE_IT(pHandle, RTC_IT_TS);
	__HAL_RTC_TIMESTAMP_GET_IT_SOURCE(pHandle, RTC_IT_TS)
	__HAL_RTC_TIMESTAMP_GET_IT(pHandle, RTC_IT_WUT)

In other families pHandle is required.
My compiler report warning: unused variable 'pHandle' [-Wunused-variable]

example:
STM32G4:
#define __HAL_RTC_WRITEPROTECTION_ENABLE(__HANDLE__)              \
  do{                                       \
    RTC->WPR = 0xFFU;   \
  } while(0)

STM32F4:
#define __HAL_RTC_WRITEPROTECTION_ENABLE(__HANDLE__)              \
    do{                                       \
       (__HANDLE__)->Instance->WPR = 0xFFU;  \
     } while(0)
Please do it always the same way

BTW
Can you explain the do {...} while(0) ?
What is the the improvment to {...}

RetoFelix

ALABSTM, 2021-02-24:
Hi @CanastraRF,

Thank you for having reported and especially for the exhaustive list of macros needing to be updated. The issue has been confirmed and logged into our internal database. A fix will be made available in a future release (**not** the next, **v1.4.0**, rather a later one).

With regards,

ALABSTM, 2021-02-24:
ST Internal Reference: 102002

ALABSTM, 2021-12-15:
Hi @CanastraRF,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.5.0** of the STM32CubeG4 published recently on GitHub. Thank you again for having reported.

Regarding your question about the advantage of surround our macros' bodies with `do { ... } while(0)` instead of simply using braces `{ ... }`, it is to cover as much cases as possible. Let me illustrate with the example below from our development teams .

Consider the following macro:
```c
#define SOME_FUNC(val) { foo(val * 2); \
                         bar(val); }
```
Assume it is used in the code snippet below:
```c
  if (...)
    SOME_FUNC(42);
  else
    ... ;
```
Because of the semicolon `;` after the `SOME_FUNC(42)` macro invocation, the code snippet will look like below once the macro expanded, which is invalid:
```c
  if (...)
  {
    foo(val * 2);
    bar(val);
  };
  else
    ... ;
```

One would argue we simply have to avoid the semicolon in this particular case. But requiring from our users to check whether a trailing semicolon is required each time they invoke a macro in their code is not the best thing to do. Hence, we use the `do { ... } while(0)` structure . I hope this answers your question.

With regards,

closed 2021-12-15 by ALABSTM

CanastraRF, 2022-01-03:
Hi @ALABSTM

Happy new Year
Thank you for the Update and the explanation

ALABSTM, 2022-01-05:
Hi @CanastraRF,

Happy new year to you too. My best wishes of health, happiness and serenity. You are more than welcome.

With regards,