STMicroelectronics/STM32CubeG4 #24: Timer Prescaler Setting Math operation will produce wrong code

VictorTagayun, 2021-02-23:
**Describe the set-up**
 * The board : NUCLEO-G474RE
 * IDE or at least the compiler and its version : CubeIDe 1.5.1 and CubeMX 6.1.1-RC2, F1.3.0
 * Win10

**Describe the bug**
TIM6 LL in CubeIDE/MX prescaler cannot use math operation like 170 - 1, (170-1), 170-2, etc.
This is not a problem with ARR.

**Screenshots**
[image]

[image]

ASELSTM, 2021-02-24:
Hi @VictorTagayun,

This issue is rather related to **CubeMX** than to the firmware published within this repository.
Actually, aspects related to CubeMX tool are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the CubeMX dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit your issue.

Since this issue is out of our scope, please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2021-02-24 by ASELSTM

VictorTagayun, 2021-03-19:
Just in case someone found this same issue.
It is a CubemX one.
https://community.st.com/s/question/0D53W00000aqkBISAY/timer-prescaler-setting-math-operation-will-produce-wrong-code