STMicroelectronics/STM32CubeG4 #25: Readme on NUCLEO-G474RE/Examples/OPAMP/OPAMP_PGA/readme.txt

VictorTagayun, 2021-02-27:
in [https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Projects/NUCLEO-G474RE/Examples/OPAMP/OPAMP_PGA/readme.txt#L51](https://github.com/STMicroelectronics/STM32CubeG4/blob/3e5243a22addb3cd323dba491eaf43c85ff448f3/Projects/NUCLEO-G474RE/Examples/OPAMP/OPAMP_PGA/readme.txt#L51)

non inverting input (PB0 - pin 34 on connector CN7) >> should be >> non inverting input (PA1 - pin 30 on connector CN7)

ASELSTM, 2021-03-02:
Hi @VictorTagayun,

Thank you for this report. Indeed, the OPAMP non inverting input pin mentioned in the readme is not coherent with the one defined in the example. It should be then updated as you have mentioned above. 
The issue you pointed is confirmed, it will be tracked internally and a fix will be implemented and made available in a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-03-02:
ST Internal Reference: 102320

ASELSTM, 2023-04-06:
Hi @VictorTagayun,

Thank you for your contribution. This issue has been fixed in the frame of version v1.5.0 of the STM32CubeG4. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM