STMicroelectronics/STM32CubeG4 #26: IS_OPAMP_POWERMODE() use legacy OPAMP_POWERMODE_NORMAL

CanastraRF, 2021-03-20:
I download STM32CubeG4 V1.4.0

In stm32g4xx_hal_opamp.h is in IS_OPAMP_POWERMODE() the legacy value OPAMP_POWERMODE_NORMAL used.

```
#define IS_OPAMP_POWERMODE(POWERMODE) (((POWERMODE) == OPAMP_POWERMODE_NORMAL) || \
                                       ((POWERMODE) == OPAMP_POWERMODE_HIGHSPEED) )

```
But the current value are:
```
#define OPAMP_POWERMODE_NORMALSPEED   (0x00000000UL)         /*!< Output in normal mode */
#define OPAMP_POWERMODE_HIGHSPEED     OPAMP_CSR_HIGHSPEEDEN  /*!< Output in highspeed mode */

```
Please fix this ASAP
ReFe

ASELSTM, 2021-03-22:
Hi @CanastraRF,

The issue you pointed out is confirmed, an internal tracker has been logged  and  a fix will be implemented and made available in the frame of a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-03-22:
ST Internal Reference: 103568

ALABSTM, 2021-12-16:
Hi @CanastraRF,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.5.0** of the STM32CubeG4 published recently on GitHub as you can see below. Thank you again for having reported.

https://github.com/STMicroelectronics/STM32CubeG4/blob/8580c58e6b0b8036d26806d43bdc115829781ab8/Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_hal_opamp.h#L337

https://github.com/STMicroelectronics/STM32CubeG4/blob/8580c58e6b0b8036d26806d43bdc115829781ab8/Drivers/STM32G4xx_HAL_Driver/Inc/Legacy/stm32_hal_legacy.h#L1006-L1007

With regards,

closed 2021-12-16 by ALABSTM

CanastraRF, 2022-01-03:
Thank You for the fix