STMicroelectronics/STM32CubeG4 #28: OPAMP

JasonLrh, 2021-09-29:
**ENV**
* arm-none-eabi-gcc 9.2.1 20191025 (release)

**Describe the bug**
When I use OPAMP3->ADC3 with PGA connect(OPAMP_PGA_GAIN_`x`_OR_MINUS_`x-1`), the output rate is `2*x-1` instead `x`
> The reference input is from ADC1

RKOUSTM, 2021-10-06:
Hi @JasonLrh,

Thank you for your contribution.

In order to allow a better analysis of this problem, could you please give us more details about the application that you are using to reproduce this issue or did you use one of our applications? Otherwise, it is possible to share with us a code snippet allowing our development teams to reproduce it.

Also, could you please confirm you are using the latest firmware version and that the issue is systematically reproducible on your side?

Thank you again for your contribution.

With regards,

RKOUSTM, 2021-12-08:
Hi @JasonLrh,

Thank you for your contribution. The point you reported is related to the CubeMX generation problem and not directly related to the firmware published in this repository. Unfortunately we don't treat aspect related to CubeMX tool in our GitHub repositories. 

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-12-08 by RKOUSTM

JasonLrh, 2021-12-08:
Hello, @RKOUSTM 

Thank you very much for your patience. After subsequent verification, the problem is that the static output of the internal operational amplifier has a certain offset, which may be related to my input environment. However, the dynamic characteristics are consistent with the set values. Your contribution to the project is very important. Thank you again for your careful answer.

Best regards