STMicroelectronics/STM32CubeG4 #29: The GNU Link Script in STM32CubeG4/Projects/NUCLEO-G491RE/Templates/STM32CubeIDE/STM32G491RETX_FLASH.ld defines wrong RAM size

Ceres-96, 2021-12-30:
The RAM size(128KB) defined in template gnu link script exceed the real size, which should be 112 KB according the g4 datasheet. This wrong parameter might cause hard fault during function call, when cpu execute the push instruction, since the sp register point to an illegal address.

`
/* Memories definition */
MEMORY
{
  CCMSRAM    (xrw)    : ORIGIN = 0x10000000,   LENGTH = 16K
  RAM    (xrw)    : ORIGIN = 0x20000000,   LENGTH = 128K
  ROM    (rx)    : ORIGIN = 0x08000000,   LENGTH = 512K
}
`

RKOUSTM, 2022-01-11:
Hi @Ceres-96,

Thank you this report. An internal bug tracker has been already opened. The fix will be made available in the frame of a future release.

Thank you once more for your report.

With regards,

RKOUSTM, 2022-01-11:
ST Internal Reference: 120645.

ALABSTM, 2023-12-25:
Hi @Ceres-96,

The issue you reported has been finally fixed and made available in the frame of release **1.5.2**. Thank you for your patience and thank you again for having reported.

With regards,

closed 2023-12-25 by ALABSTM