STMicroelectronics/STM32CubeG4 #30: missing static const in static CRYP_AESGCM_Process_IT()

CanastraRF, 2022-01-03:
Hi 
In the newest version of 
`static HAL_StatusTypeDef CRYP_AESGCM_Process_IT(CRYP_HandleTypeDef * const hcryp)
`
is a mask array
 ```
 uint32_t mask[12U] = {0x0U, 0xFF000000UL, 0xFFFF0000UL, 0xFFFFFF00UL,  /* 32-bit data type */
                       0x0U, 0x0000FF00UL, 0x0000FFFFUL, 0xFF00FFFFUL,  /* 16-bit data type */
                       0x0U, 0x000000FFUL, 0x0000FFFFUL, 0x00FFFFFFUL}; /*  8-bit data type */
```

this should be static const.
Please fix this.

RKOUSTM, 2022-01-07:
Hi @CanastraRF,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-01-07:
ST Internal Reference: 120501

RJMSTM, 2023-07-31:
Fixed in commit c356ba51f9673113cee29414857ad019c213588a

closed 2023-07-31 by RJMSTM