STMicroelectronics/STM32CubeG4 #31: Why PDM2PCM is not added in G4?

VictorTagayun, 2022-04-04:
I added manually PDM2PCM in G4, so far it is working.
Why is it not added in G4?
I tried to ask in the community https://community.st.com/s/question/0D53W00001TGJ2wSAH/why-is-pdm2pcm-not-enabled-for-stm32g4-in-cubemx

RKOUSTM, 2022-04-11:
Hi @VictorTagayun,

First, allow me to thank you for your report. However, the point you reported is not directly related to the firmware published in this repository and is out of the scope of STMicroelectronics' activity. They are rather treated at [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level. 

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to a user application, please allow me to close it.

Thank you for your comprehension. We are looking forward to reading from you again.

With regards,

closed 2022-04-11 by RKOUSTM

VictorTagayun, 2022-04-11:
Why you cannot add PDM2PCM in G4?