STMicroelectronics/STM32CubeG4 #37: Missing package definitions

carlo-dev-git, 2023-03-10:
In file `stm32g4xx_ll_utils.h` the package type list should be fixed:
```
/** @defgroup UTILS_EC_PACKAGETYPE PACKAGE TYPE
  * @{
  */
#define LL_UTILS_PACKAGETYPE_LQFP64         0x00000000U /*!< LQFP64 package type                      */
#define LL_UTILS_PACKAGETYPE_LQFP80 <-- MISSING !!!
#define LL_UTILS_PACKAGETYPE_LQFP100        0x00000002U /*!< LQFP100 package type                     */
#define LL_UTILS_PACKAGETYPE_WLCSP81        0x00000005U /*!< WLCSP81 package type                     */
#define LL_UTILS_PACKAGETYPE_LQFP128        0x00000007U /*!< LQFP128 package type                     */
#define LL_UTILS_PACKAGETYPE_UFQFPN32       0x00000008U /*!< UFQFPN32 package type                    */
#define LL_UTILS_PACKAGETYPE_LQFP32         0x00000009U /*!< LQFP32 package type                      */
#define LL_UTILS_PACKAGETYPE_UFQFPN48       0x0000000AU /*!< UFQFPN48 package type                    */
#define LL_UTILS_PACKAGETYPE_LQFP48         0x0000000BU /*!< LQFP48 package type                      */
#define LL_UTILS_PACKAGETYPE_WLCSP49        0x0000000CU /*!< WLCSP49 package type                     */
#define LL_UTILS_PACKAGETYPE_UFBGA64        0x0000000DU /*!< UFBGA64 package type                     */
#define LL_UTILS_PACKAGETYPE_UFBGA100 <-- not exists... TFBG100 maybe?
#define LL_UTILS_PACKAGETYPE_UFBGA121 <-- MISSING !!!
#define LL_UTILS_PACKAGETYPE_LQFP48_EBIKE   0x00000010U /*!< LQFP48 EBIKE package type                */
```

TOUNSTM, 2023-07-27:
ST Internal Reference: 158200

RJMSTM, 2023-11-02:
Fixed in commit 7836572c2150d03adf0f9afdacd7c1a3077f748b

closed 2023-11-02 by RJMSTM