STMicroelectronics/STM32CubeG4 #40: HAL_RTC_Init() loses time

ay2718, 2023-06-20:
Same issue as https://github.com/STMicroelectronics/STM32CubeH7/issues/24 and https://github.com/STMicroelectronics/STM32CubeF1/issues/58.  HAL_RTC_Init() is called on each reset which loses up to a second of time.

RJMSTM, 2023-06-22:
Hello @ay2718,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,
Rania

HBOSTM, 2023-06-23:
ST Internal Reference: 126470

HBOSTM, 2023-06-23:
Hello @ay2718 ,

Thank you for this contribution. This problem has been fixed in the [stm32g4xx_hal_driver](https://github.com/STMicroelectronics/stm32g4xx_hal_driver/commit/11b72c0a7ebcb8fc315cb7df32550b317f9eefff), it will be available in the next G4-cube Firmware release  (**v1.5.2**).

Best Regards,

RJMSTM, 2023-07-31:
Fixed in commit c356ba51f9673113cee29414857ad019c213588a

closed 2023-07-31 by RJMSTM