STMicroelectronics/STM32CubeG4 #41: template file system_stm32g4xx.c should support userdefined location of vectortable

marcgendron-dyze, 2023-08-09:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
 * IDE or at least the compiler and its version.

**Describe the bug**
Same as https://github.com/STMicroelectronics/STM32CubeF4/issues/30
`A clear and concise description of what the bug is.
template file system_stm32g4xx.c should support userdefined location of vectortable
using a custom bootloader requires to move the default location of the vector table.
in the file system_stm32g4xx.c that is copied by the program generator is the needed settings for the move of the vector table.
But the user can not control this behaviour by compiler command arguments
in the defines section at the beginning of the file line 92ff
the lines:
/* #define VECT_TAB_SRAM /
#define VECT_TAB_OFFSET 0x00 /!< Vector Table base offset field.
This value must be a multiple of 0x200. /
/*****************************************************************************/

have to be changed to
/* #define VECT_TAB_SRAM /
#if !defined(VECT_TAB_OFFSET)
#define VECT_TAB_OFFSET 0x00 /!< Vector Table base offset field.
This value must be a multiple of 0x200. /
#endif
/*****************************************************************************/

So the user can control the define from compiler command arguments.
This allows targets for bootloader or plain memory layout.

All the existing code will not break.`

**How To Reproduce**
1. Indicate the global behavior of your application project.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).

3. The use case that generates the problem.

4. How we can reproduce the problem.

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2023-08-18:
Hi @marcgendron-dyze,

Thank you for this report. We are already aware of this point and are working on a global update. We will let you know once a fix is published. We cannot share a date for the moment. Thank you for your comprehension.

With regards,

ALABSTM, 2023-11-29:
ST Internal Reference: 116475

ALABSTM, 2025-02-21:
Fixed in commit 2478bddd92bea5d0d45e4b8e85ef7f5c64246a3d

ALABSTM, 2025-02-21:
Hi @marcgendron-dyze,

I hope you are doing well. We finally updated the [system_stm32g4xx.c](https://github.com/STMicroelectronics/cmsis-device-g4/blob/b7047a31b98145d930f26a3144985bb845783a6c/Source/Templates/system_stm32g4xx.c#L123-L126) file to allow macro `VECT_TAB_OFFSET` to be overridden externally.

Thank you for your patience and thank you again for having reported.

With regards,

closed 2025-02-21 by ALABSTM