STMicroelectronics/STM32CubeG4 #46: Obsolete type caddr_t

doodeca, 2023-10-27:
When building STM32CubeG4 examples using STM32CubeIDE Version 1.13.2, I get the following error:
'unknown type name 'caddr_t' ' in `[example]/STM32CubeIDE/Application/User/sysmem.c:63:1`

Including `<sys/types.h>` fixes the error. However, `caddr_t` is [an obsolete typedef](https://stackoverflow.com/questions/6381526/what-is-the-significance-of-caddr-t-and-when-is-it-used). It's recommended to replace these with `void*`.

TOUNSTM, 2023-12-14:
Hello @doodeca,

Thank you for your report. In order to support GCC11, we need to update the sysmem.c and syscall.c files in all STM32CubeIDE projects. The correction will be available in the next release..

With regards,
Tasnim

TOUNSTM, 2023-12-14:
Hello @doodeca,

Thank you for your report. In order to support GCC11, we need to update the sysmem.c and syscall.c files in all STM32CubeIDE projects. The fix will be available in the next release.

With regards,

ALABSTM, 2023-12-25:
Hi @doodeca,

Issue fixed in the frame of release **1.5.2**. Thank you for your report.

With regards,

closed 2023-12-25 by ALABSTM