STMicroelectronics/STM32CubeG4 #47: EXTI_ToggleLedOnIT and EXTI_ToggleLedOnIT_Init are totally the same

VictorTagayun, 2023-12-02:
Documentation issues:

1. EXTI_ToggleLedOnIT and EXTI_ToggleLedOnIT_Init are totally the same, both main.c and ioc files
2. It mentions in the readme file that it is triggered at the falling edge but it should be rising edge as shown in its SCH. 
[image]
3. The ioc files labels PC4 as USER_BUTTON, while it should be PC13
[image]
4.  PC4 is not used and can be Reset_State, it is here that the external trigger is falling edge.

TOUNSTM, 2023-12-14:
Hello @VictorTagayun,

We confirm the abnormal behavior you reported. It has been reported to our development teams for deeper analysis. We will get back to you as soon we have updates.

With regards,

TOUNSTM, 2023-12-21:
ST Internal Reference: 169451

TOUNSTM, 2024-07-02:
Fixed in 98f52ec7cdbbbfe1650b351bde2efb037815886d

closed 2024-07-02 by TOUNSTM