STMicroelectronics/STM32CubeG4 #49: __PKHBT and __PKHTB are STILL wrong

etheory, 2024-03-27:
As you can see here: https://github.com/STMicroelectronics/STM32CubeG4/blob/29e23e9f45f0a495eafbfb341d98d65ca967e6ff/Drivers/CMSIS/Include/cmsis_gcc.h#L2129-L2152

The version in this repository is STILL incorrect, and should be fixed please.

I reported this months ago and you closed it since you said it didn't match the CMSIS version, but clearly it does.

RJMSTM, 2024-03-29:
Hello @etheory,

The latest version of CMSIS available on the STM32CubeG4 repository is 5.6.0, which does not include the most recent commits indicating that this change has already been made."

[image]

I will forward this matter to the development team to determine whether we will integrate the latest version of CMSIS.

Thank you for your contribution.

Regards,

etheory, 2024-03-29:
> Hello @etheory,
>
> The latest version of CMSIS available on the STM32CubeG4 repository is 5.6.0, which does not include the most recent commits indicating that this change has already been made." [...]

@RJMSTM CMSIS-5.6.0 is 5 years old. The commit that resolved this issue is 2 years old. That's pretty old. It definitely feels like your dependencies should be more up to date than that. Thanks.

KRASTM, 2024-10-09:
Hello @etheory,

Thank you for the report.

Regarding your issue, indeed the version of CMSIS available on STM32G4 is v5.6.0 and it's not synchronized with ARM version, but unfortunately there is no plan to upgrade. I will keep you informed if there are any updates. Please allow me to close this issue.

Thank you in advance for your comprehension.

With regards,

etheory, 2024-10-09:
@KRASTM v5.6.0 is now 5 years old. Why are you so out of date with your library dependencies?
By all means close the ticket, but you are costing your customers significant runtime performance by keeping the library dependencies so old.

KRASTM, 2024-10-10:
Hi @etheory,

It is not the case, and this is not the purpose behind our support, but sometimes due to some factors we confirm the issue, but we do not have a forecast or a plan to fix it, decisions are made by our team.

Additionally, there are other series up to date with ARM and others not, for example, U5 or H5 with V5.9.0

Thank you again.
With regards,

closed 2024-10-10 by KRASTM

etheory, 2024-10-10:
Thanks @KRASTM but why all the inconsistency, why not just support the same version across all chips? It's super confusing for your users, and adds complexity that isn't necessary.

Are you open to open source contributions? It should be relatively easy to fix such a thing (I'm an applications c++ programmer by trade, I'd be happy to help improve the libraries).