STMicroelectronics/STM32CubeG4 #52: Reduce ROM size of USB_EPStartXfer

KozhinovAlexander, 2024-12-02:
This issue were observed within zephyr rtos project using stm32g4, USB stack from zephyr rtos and stm32g4 HAL drivers.

**Is your feature request related to a problem? Please describe.**
`USB_EPStartXfer` for stm32g4 has 1440 KiBi text section, which is huge in my opinion. 

**Describe the solution you'd like**
The size of `USB_EPStartXfer`  shall be reduced or the reason for such text section size shall be mentioned.

**Describe alternatives you've considered**
none

**Additional context**
To reproduce the behaviour any USB sample for stm32g4 could be taken. Afterwards execute `ninja -C build rom_report > rom_report.log`.

In my case I am getting following values:

```
ninja: Entering directory `app/build'
[0/1] cd $HOME/Documents/dev/cannectivity/app/build &&  $HOME/Documents/dev/zephyrproject/.zephyr_venv/bin/python3  $HOME/Documents/dev/zephyrproject/zephyr/scripts/footprint/size_report -k  $HOME/Documents/dev/cannectivity/app/build/zephyr/zephyr.elf -z $HOME/Documents/dev/zephyrproject/zephyr -o  $HOME/Documents/dev/cannectivity/app/build --workspace= $HOME/Documents/dev/zephyrproject -d 99 rom
WARN: Symbol 'z_tls_current' section '' is not in RAM or ROM.
Path                                                                                             Size       %  Address    Section    
==========================================================================================================================================
Root                                                                                            61146 100.00%  - 
├── (hidden)                                                                                     8088  13.23%  - 
├── (no paths)                                                                                  53056  86.77%  - 
│   ├── AHBPrescTable                                                                              16   0.03%  0x0800defd rodata
│   ├── APBPrescTable                                                                               8   0.01%  0x0800def5 rodata
│   ├── CSWTCH.490                                                                                 20   0.03%  0x0800bde8 rodata
│   ├── HAL_PCD_DataInStageCallback.constprop.0                                                    36   0.06%  0x0800b721 text
│   ├── HAL_PCD_DataOutStageCallback.constprop.0                                                   56   0.09%  0x08007579 text
│   ├── HAL_PCD_SuspendCallback.constprop.0                                                        20   0.03%  0x080075b1 text
│   ├── LL_RCC_HSI_IsReady                                                                         16   0.03%  0x08005ee5 text
│   ├── LL_SetFlashLatency                                                                        220   0.36%  0x08006031 text
│   ├── LL_TIM_ClearFlag_CC1                                                                        8   0.01%  0x0800a805 text
│   ├── LL_TIM_ClearFlag_CC2                                                                        8   0.01%  0x0800a80d text
│   ├── LL_TIM_ClearFlag_CC3                                                                        8   0.01%  0x0800a815 text
│   ├── LL_TIM_ClearFlag_CC4                                                                        8   0.01%  0x0800a81d text
│   ├── LL_TIM_DisableIT_CC1                                                                       10   0.02%  0x0800a82f text
│   ├── LL_TIM_DisableIT_CC2                                                                       10   0.02%  0x0800a843 text
│   ├── LL_TIM_DisableIT_CC3                                                                       10   0.02%  0x0800a857 text
│   ├── LL_TIM_DisableIT_CC4                                                                       10   0.02%  0x0800a86b text
│   ├── LL_TIM_EnableIT_CC1                                                                        10   0.02%  0x0800a825 text
│   ├── LL_TIM_EnableIT_CC2                                                                        10   0.02%  0x0800a839 text
│   ├── LL_TIM_EnableIT_CC3                                                                        10   0.02%  0x0800a84d text
│   ├── LL_TIM_EnableIT_CC4                                                                        10   0.02%  0x0800a861 text
│   ├── LL_TIM_OC_GetCompareCH1                                                                     4   0.01%  0x0800a7f5 text
│   ├── LL_TIM_OC_GetCompareCH2                                                                     4   0.01%  0x0800a7f9 text
│   ├── LL_TIM_OC_GetCompareCH3                                                                     4   0.01%  0x0800a7fd text
│   ├── LL_TIM_OC_GetCompareCH4                                                                     4   0.01%  0x0800a801 text
│   ├── LL_TIM_OC_SetCompareCH1                                                                     4   0.01%  0x0800a7e5 text
│   ├── LL_TIM_OC_SetCompareCH2                                                                     4   0.01%  0x0800a7e9 text
│   ├── LL_TIM_OC_SetCompareCH3                                                                     4   0.01%  0x0800a7ed text
│   ├── LL_TIM_OC_SetCompareCH4                                                                     4   0.01%  0x0800a7f1 text
│   ├── SEGGER_RTT_WriteSkipNoLock.constprop.0                                                    112   0.18%  0x0800760d text
│   ├── SystemCoreClock                                                                             4   0.01%  0x20000f1c datas
│   ├── USB_ActivateEndpoint                                                                      776   1.27%  0x08006cb5 text
│   ├── USB_EPStartXfer                                                                          1440   2.36%  0x0800ae4d text
```

especially:

```
│   ├── USB_EPStartXfer                                                                          1440   2.36%  0x0800ae4d text
```

ALABSTM, 2024-12-20:
Hi @KozhinovAlexander,

Thank you for this report. From my side, using EWARM 9.20.4 IDE, I found the following values:
* a size equal to **0x36C** bytes (i.e., **876** bytes) with a high "balanced" optimization,
* a size equal to **0x2C0** bytes (i.e., **704** bytes) with a high "size" optimization,
* a size equal to **0x38E** bytes (i.e., **910** bytes) with a high "speed" optimization,

I will forward your request to our development teams. I will let you know once I have their feedback.

With regards,

ALABSTM, 2024-12-20:
Hi @KozhinovAlexander,

By the way, may I know the optimization level you have used to obtain the figures you shared?

Thank you,

KozhinovAlexander, 2024-12-20:
@ALABSTM Thank you for taking l a look on it.
Here you can find .conf file, which has in the first three lines optimizations:https://github.com/CANnectivity/cannectivity/pull/52

KozhinovAlexander, 2024-12-20:
@ALABSTM Thank you for asking more information. I'll try to ptovide as much as possible:

Here are compile commands listed:

1. It shall be -Os:
```json
"directory": "<build_dir_here>",

"command": "$HOME/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DSTM32G431xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I$HOME/Documents/dev/cannectivity/app/build/zephyr/include/generated/zephyr -I$HOME/Documents/dev/zephyrproject/zephyr/include -I$HOME/Documents/dev/cannectivity/app/build/zephyr/include/generated -I$HOME/Documents/dev/zephyrproject/zephyr/soc/st/stm32 -I$HOME/Documents/dev/zephyrproject/zephyr/soc/st/stm32/common/. -I$HOME/Documents/dev/zephyrproject/zephyr/drivers -I$HOME/Documents/dev/zephyrproject/zephyr/soc/st/stm32/stm32g4x/. -I$HOME/Documents/dev/zephyrproject/zephyr/subsys/usb/device -I$HOME/Documents/dev/zephyrproject/custom/cannectivity/include -I$HOME/Documents/dev/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I$HOME/Documents/dev/zephyrproject/zephyr/modules/cmsis/. -I$HOME/Documents/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32g4xx/soc -I$HOME/Documents/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32g4xx/drivers/include -I$HOME/Documents/dev/zephyrproject/modules/hal/stm32/stm32cube/common_ll/include -I$HOME/Documents/dev/zephyrproject/modules/debug/segger/SEGGER -I$HOME/Documents/dev/zephyrproject/modules/debug/segger/Config -isystem $HOME/Documents/dev/zephyrproject/zephyr/lib/libc/common/include -fno-strict-aliasing -Os -flto=auto -imacros $HOME/Documents/dev/cannectivity/app/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=$HOME/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros $HOME/Documents/dev/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=$HOME/Documents/dev/cannectivity/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=$HOME/Documents/dev/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=$HOME/Documents/dev/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=picolibc.specs -std=c99 -o modules/hal_stm32/stm32cube/CMakeFiles/..__modules__hal__stm32__stm32cube.dir/stm32g4xx/drivers/src/stm32g4xx_ll_usb.c.obj -c $HOME/Documents/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32g4xx/drivers/src/stm32g4xx_ll_usb.c",

"file": "$HOME/Documents/dev/zephyrproject/modules/hal/stm32/stm32cube/stm32g4xx/drivers/src/stm32g4xx_ll_usb.c",

"output": "modules/hal_stm32/stm32cube/CMakeFiles/..__modules__hal__stm32__stm32cube.dir/stm32g4xx/drivers/src/stm32g4xx_ll_usb.c.obj"
```

Here I get: `│   ├── USB_EPStartXfer                                                                           988   1.53%  0x0800b79d text`

3. tbd
4. tbd

ALABSTM, 2024-12-23:
ST Internal Reference: 199294

ALABSTM, 2025-01-21:
Hi @KozhinovAlexander,

I hope you are doing well. According to development teams, the symbol `USE_USB_DOUBLE_BUFFER` should be defined to `0`. This reduces considerably the size of the `USB_EPStartXfer()` function, provided you do not need the double buffering feature.

From my side, still using EWARM 9.20.4 IDE, I set `USE_USB_DOUBLE_BUFFER=0` in the preprocessor symbols and built [this](https://github.com/STMicroelectronics/STM32CubeG4/tree/master/Projects/STM32G474E-EVAL1/Applications/USB_Device/CDC_Standalone) application with different optimization levels. I found the following values:

* A size reduced from **0x36C** bytes  (i.e., **876** bytes) to **0xC8** bytes (i.e., **200** bytes) with a high "balanced" optimization, corresponding to a **77.2%** footprint reduction.
* A size reduced from **0x2C0** bytes (i.e., **704** bytes) to **0xAC** bytes (i.e., **172** bytes) with a high "size" optimization, corresponding to a **75.5%** footprint reduction.
* A size reduced from **0x38E** bytes (i.e., **910** bytes) to **0x112** bytes (i.e., **274** bytes) with a high "speed" optimization, corresponding to a **69.9%** footprint reduction.

I hope this addresses your needs. Please let me know your feedback.

With regards,

**PS:** Best wishes for this new year.

KozhinovAlexander, 2025-02-07:
> Hi [@KozhinovAlexander](https://github.com/KozhinovAlexander),
>
> I hope you are doing well. According to development teams, the symbol `USE_USB_DOUBLE_BUFFER` should be defined to `0`. This reduces [...]

It took me some time to reply. :) Your solution is perfect and frees a lot of memory now: 

```
│   ├── USB_EPStartXfer                                                                           216   0.38%  0x0800a215 text
```

So I got 216 bytes instead of 1440, which is really great.
I need to add that I've tested slightly newer firmware, where one of USB endpoints were dropped, since dummy. But the achievement is great. I am happy with it. Please confirm as solved. Thank you a lot.

closed 2025-02-07 by KozhinovAlexander

ALABSTM, 2025-03-17:
Fixed in commit eee918145c5a9a891a3389c71a50691d6ff597d5

ALABSTM, 2025-03-17:
Hi @KozhinovAlexander,

Just to let you know we added a [note](https://github.com/STMicroelectronics/stm32g4xx-hal-driver/commit/69accade0a117e66abbd6620ac95ebbc3de0f4d8) to inform about this possibility of code footprint optimization using the symbol `USE_USB_DOUBLE_BUFFER`.

KozhinovAlexander, 2025-03-17:
> Hi @KozhinovAlexander,
> 
> Just to let you know we added a [note](https://github.com/STMicroelectronics/stm32g4xx-hal-driver/commit/69accade0a117e66abbd6620ac95ebbc3de0f4d8) to inform about this possibility of **code footprint optimization** using the symbol `USE_USB_DOUBLE_BUFFER`.

You're great. Thank you.