STMicroelectronics/STM32CubeG4 #53: Macro LL_ADC_SetChannelSingleDiff behaves incorrectly when setting a channel to single ended

microbug, 2025-01-06:
**Describe the set-up**
- Followed [this guide on the ST Wiki](https://wiki.st.com/stm32mcu/wiki/STM32StepByStep:Getting_started_with_USB-Power_Delivery_Sink#Enable_the_software_pack) to set up Nucleo with USB Type-C PD sink.
- Boards: Nucleo-G474RE with X-NUCLEO-SNK1M1 USB Type-C PD sink daughterboard.
- STM32CubeIDE v1.17.0

**Describe the bug**
The macro function `LL_ADC_SetChannelSingleDiff` should clear bit `ADCx->DIFSEL[ 1 << n ]` when setting channel n to single-ended. Instead, the bit is set. This causes the channel to be incorrectly set to differential, resulting in invalid measurements.

See STM32G4 Reference Manual, RM0440 Rev 8 section 21.7.21.

**How To Reproduce**
1. Indicate the global behavior of your application project.
When running the application, ADC readings are incorrect. This makes it impossible to verify that the USB Type-C PD VBUS voltage is safe.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
[This macro function in HAL-LL Drivers](https://github.com/STMicroelectronics/stm32g4xx-hal-driver/blob/6cdb4c3f4d485eb2bd0717fa240f3ff9687f96dc/Inc/stm32g4xx_ll_adc.h#L6490C22-L6490C49).

3. The use case that generates the problem.
Generate default project following link below, and measure VBUS voltage by setting a breakpoint in the relevant function.

4. How we can reproduce the problem.
Generate a project following [this guide on the ST Wiki](https://wiki.st.com/stm32mcu/wiki/STM32StepByStep:Getting_started_with_USB-Power_Delivery_Sink#Enable_the_software_pack) using Nucleo-G474RE and X-NUCLEO-SNK1M1 boards.

Run project in Debug mode. Place a breakpoint on line `*pVoltage = voltage` in function `BSP_USBPD_PWR_VBUSGetVoltage`, file `USBPD/usbpd_pwr_user.c`. Observe that when a USB charger is plugged into the USB Type-C port, the voltage calculated does not match expectations.

Because the ADC is running in continuous mode, you can also stop execution and inspect register ADC1->DR. When no USB Type-C device is plugged in, it should be close to 0x0. It will instead be much higher.

Also observe that when placing a breakpoint on the call to `LL_ADC_SetChannelSingleDiff` in file `Drivers/STM32G4xx_HAL_Driver/Src/stm32g4xx_hal_adc.c` (line 2885 in attached project), ADC1->DIFSEL is 0x0 before the call and 0x8000 after the call (bit 15 has been set).

**Additional context**
The macro currently looks like this (comment is not mine):

```c
__STATIC_INLINE void LL_ADC_SetChannelSingleDiff(ADC_TypeDef *ADCx, uint32_t Channel, uint32_t SingleDiff)
{
  /* Bits of channels in single or differential mode are set only for         */
  /* differential mode (for single mode, mask of bits allowed to be set is    */
  /* shifted out of range of bits of channels in single or differential mode. */
  MODIFY_REG(ADCx->DIFSEL,
             Channel & ADC_SINGLEDIFF_CHANNEL_MASK,
             (Channel & ADC_SINGLEDIFF_CHANNEL_MASK)
             & (ADC_DIFSEL_DIFSEL >> (SingleDiff & ADC_SINGLEDIFF_CHANNEL_SHIFT_MASK)));
}
```

`Channel` is `(1 << n)` where `n` is channel number, plus some other stuff beyond bit 24, which is masked out by `ADC_SINGLEDIFF_CHANNEL_MASK`
`ADC_SINGLEDIFF_CHANNEL_MASK` is 0x7FFFF
`ADC_DIFSEL_DIFSEL` is 0x7FFFF
`ADC_SINGLEDIFF_CHANNEL_SHIFT_MASK` is 24

So the simplified expansion is:
```c
ADCx->DIFSEL = ADCx->DIFSEL & ~(1 << n) | ( (1 << n) & (0x7FFFF >> (SingleDiff & 24)) )
```

In the case where n=15 and SingleDiff=0 (which should set the channel to single-ended), this expands to:
```c
ADCx->DIFSEL = ADCx->DIFSEL & ~(0x8000) | ( 0x8000 & 0x7FFFF )
```
which expands to:
```c
ADCx->DIFSEL = ADCx->DIFSEL & ~(0x8000) | 0x8000
```
Bit 15 is set by this macro.

**Screenshots**
None.

**Project**
Demo project files are attached: [stm32g4xx_adc_bug.zip](https://github.com/user-attachments/files/18318445/stm32g4xx_adc_bug.zip).

RJMSTM, 2025-01-10:
Hellon @microbug,

First thanks for your contribution, i am trying to analyze the code you sent in the zip file, but I am having trouble. 

[image]

Could you please specify which part of your code is causing the ADC issue 'when setting channel n to single-ended. Instead, the bit is set. This causes the channel to be incorrectly set to differential'?

What is the version of STM32CubeIDE being used?

Regards,

microbug, 2025-01-10:
The cause of the problem is file `Drivers/STM32G4xx_HAL_Driver/Inc/stm32g4xx_ll_adc.h` lines 6495-6498. When `SingleDiff` is 0, `DIFSEL[n]` should be set to 0 according to the reference manual (`n` is the channel number). This sets the channel to single ended.

However, this macro call does not do that, as explained above. Instead it sets `DIFSEL[n]` to 1, which causes incorrect measurements.

STM32CubeIDE is version 1.17.0, build 23558_20241125_2245 (UTC).

RJMSTM, 2025-01-23:
Hello @microbug,

I don't think the cause of the problem is in our driver. I tried testing this function with different channels. For example, I used channel 1 and channel 16, and I set both configurations to `SingleDiff `and `Differentended`. The function runs fine.

**1. Channel 1, `Differentended`**

[image]

**2.  Channel 1, `SingleDiff `**

[image]

**3. Channel 16, `Differentended`**

[image]

**4. Channel 16, `SingleDiff`** 

[image]

As you have seen in the screenshots, in the DIFSEL[n] register, set it to 0 if SingleDiff is selected and set it to 1 if Differentended is selected for channel n. This is illustrated with examples for channels 1 and 16.

Tested with: 
STM32CubeIDE is version 1.17.0
STM32G474

Regards,

RJMSTM, 2025-02-03:
Hello @microbug,

If everything is clear to you, may I close the issue? You can reopen it if you have any questions or problems.

Rgards,

closed 2025-02-07 by ALABSTM